Caffreyfans/IRbaby #7: mqtt断线重连问题

yuqiuyi99, 2019-05-13:
经几次验证，正常使用中，如果hass重启，mqtt连接会断开。虽然nodemcu会一直尝试重连，但不成功。rx口输出如下：
[12:55:27.196]收←◆MQTT try to reconnect
[Connecting to MQTT]: 
[12:55:33.856]收←◆. 
[12:55:57.196]收←◆MQTT try to reconnect
[Connecting to MQTT]: . 
[12:56:03.862]收←◆. 
但如果重启nodemcu，则会立即连上mqtt。

是否可以优化mqtt断线重连机制？
或者是否可以在固件中加入一个重启命令，像espeasy一样输入一个命令就可以让nodemcu重启，例如espdasy的指令是：http://192.168.1.xx/?cmd=reboot；sonoff命令是http://192.168.1.xx/cm?cmnd=Restart 1，这样也可以在hass重启后发送命令重置重启nodemcu。

Caffreyfans, 2019-05-13:
固件现在是的连接机制是：如果 mqtt 断开连接，就每隔 30 秒重新连接一次。断开连接分两种情况，一是服务端停了导致 mqtt 连接不上，如果服务器正常了就会再连上去，二是连接参数有问题导致连接不上服务器，这种情况只有重新配置正确的参数才会连接上，这种情况也就包括了还没填上 mqtt 连接参数的时候。我现在也没想到什么好的优化方法，顶多再判断下连接参数是否为空，如果这种情况就不重新连接了，你还有什么优化方案吗？分享一下

yuqiuyi99, 2019-05-14:
在树莓派+debian+docker+hassio下，重启hass后固件连不上。不知道你说的第一种情况为什么没有实现。但此时重启nodemcu就会立即连接上原来的mqtt服务器。不知是否是个案还是通病。
  如果这种现象具有普遍性，那么：现有机制是如果发现断开mqtt，就每隔 30 秒重新连接一次。是否能在连接机制里再加入一个600s的计时器，断开mqtt600s仍连不上mqtt服务器就rst

Caffreyfans, 2019-05-22:
代码我改了，本来像让模块在不断电的情况下重新连接到 MQTT 服务器，不知道为什么始终不行。后面再看看到底是怎么回事吧，模块重启会让空调又恢复到默认状态所以并不是很好的方式。现阶段我也没找到原因，就暂时使用重启的方式让它恢复连接吧

closed 2019-05-22 by Caffreyfans