Caffreyfans/IRbaby #9: 求继续维护IRmqtt-tool.py

theosoft-git, 2020-03-01:
照顾下mac用户的感受吧！刷了2.2的bin，然后去源码里找了IRmqtt-tool.py，但是每次“匹配电器”都直接报错退出。附错误日志：
【主菜单】
[1] 设置红外发射引脚
[2] MQTT 连接设置
[3] 匹配电器(当前只支持空调)
[4] 设置红外接收引脚
[5] 录制自定义红外码
[6] 退出程序
请输入选项: 3
# # # # # # # # # # # # # # # # # # # # # # # #
Traceback (most recent call last):
  File "./IRmqtt-tool.py", line 537, in <module>
    main()
  File "./IRmqtt-tool.py", line 533, in main
    menu = Menu()
  File "./IRmqtt-tool.py", line 477, in __init__
    self.run(self._choice_dict, self._choice_dict)
  File "./IRmqtt-tool.py", line 523, in run
    v()
  File "./IRmqtt-tool.py", line 463, in parse_ac
    irext = Irext(self._device_name, self._device_ip)
  File "./IRmqtt-tool.py", line 86, in __init__
    '4f9697c87d75ebaf2522234c72410398')
  File "./IRmqtt-tool.py", line 186, in app_login
    auther = {'id':ret_json['entity']['id'], 'token':ret_json['entity']['token'] }
KeyError: 'entity'

Caffreyfans, 2020-03-03:
马上

Caffreyfans, 2020-03-03:
这个是我的失误，这个和 MAC 没关系。由于数据是从 irext 获取的，它需要 SDK 账号认证。有一天不消息把账号删除了，没及时更新。抱歉

Caffreyfans, 2020-03-04:
# 更新了，现在分离了。python 的客户端专门在 [image] 仓库里了

theosoft-git, 2020-03-06:
嗯，试了下。现在果然可以运行了。但是我家的日立中央空空调，日立下面9个遥控竟然都没匹配上？！

Caffreyfans, 2020-03-13:
你先确认下，红外信号发射没有。如果发射了空调还是没反映再说下，我好向码库作者说明下情况。

mibqooqoo, 2020-03-13:
匹配空調 有反應 但是照內容寫入Ha後 無法控制 有設定MQTT 也有在Ha設定MQTT hub
[image]

Caffreyfans, 2020-03-27:
> 匹配空調 有反應 但是照內容寫入Ha後 無法控制 有設定MQTT 也有在Ha設定MQTT hub
> [image]

麻烦再确认一下，因为你已经说了。匹配空调的时候有反映，就是 HA 里无法控制，HA 里是通过 MQTT 通信来控制，所以确认下 MQTT 相关配置是否正确。

closed 2020-04-27 by Caffreyfans