Caffreyfans/IRbaby #11: app里家电控制页添加空调设备时可控，添加后不可控

ghostist, 2020-05-23:
问题描述：
app里家电控制页添加空调设备，选择长虹3或长虹13，空调有响应；
确定选择后，再进行控制，则空调能接收到指令（表现为滴滴滴响），状态始终为17°，模式为最后一次设定的模式。

环境：
win10，雷电模拟器，IRbaby 安卓客户端0.0.3，IRbaby 固件0.0.3
X12移动空调伴侣，模块替换ESP8266（4Mflash）,设置14脚发送 13脚接收，mqtt设置且已连接

复现：
如上述环境设置，app左侧“家电控制”页，找到局域网设备，添加空调-长虹3或长虹13
PS：我家空调只有制热、制冷模式生效，其他模式空调不响应，空调型号：KFR-35GW/DHR(W1-H)+2
在点击下方的【是】之前，空调响应指令，且正确
点击【是】后重新进入该空调控制界面，任意指令发送，空调均响应为17°，模式不变

其他：
日志中，选择设备和确认设备后的控制页面，同样的状态下（模式、风速、风向、扫风、温度等），日志输出
{
  "params": {
    "set": "set",
    "file": "new_ac_10707",
    "var": "{\"power\":0,\"temperature\":10,\"mode\":2,\"swing\":1,\"direction\":1,\"speed\":0}"
  }
}
中的file字段都一致，但后面DEBUG: length = 243具体的指令码有区别

icandura, 2020-08-07:
我这里也出现了这个问题，在添加空调的时候，在添加页面的“格力 25”可以完美控制，但是点击“是”添加为正式的空调之后，就不能控制了。

Caffreyfans, 2020-08-07:
> 我这里也出现了这个问题，在添加空调的时候，在添加页面的“格力 25”可以完美控制，但是点击“是”添加为正式的空调之后，就不能控制了。

你加下我的 QQ，我好详细追踪下这个问题。我好立即修复，因为自己没有测试环境。

icandura, 2020-08-08:
> > 我这里也出现了这个问题，在添加空调的时候，在添加页面的“格力 25”可以完美控制，但是点击“是”添加为正式的空调之后，就不能控制了。
> 
> 你加下我的 QQ：893399065，我好详细追踪下这个问题。我好立即修复，因为自己没有测试环境。

我后面猜测可能是这样，正如同我之前反馈的另外的一个问题。我刷的是0.4的固件，手机上用0.4版本的APP找不到设备，反而用0.3版本可以，所以就是在0.3版本的APP上控制0.4固件的8266，出现了这个试码过程中可以控制，但添加后就不能控制的情况。
后来我把固件也降级到0.31的版本之后，再用0.3版本的APP来控制，就没有这个问题了。。

Caffreyfans, 2020-08-12:
问题已在 0.5 版本修复.

closed 2020-08-12 by Caffreyfans