Caffreyfans/IRbaby #18: 新的安卓客户端搜索不到局域网的8266

icandura, 2020-08-07:
0.4版本的安卓客户端安装后，搜索不到局域网内刷了固件的8266。
后来卸载掉，安装了旧版本（0.3版本）的客户端，一进去就能找到了。

Caffreyfans, 2020-08-07:
新的客户端要配合新的固件哦，现在协议不稳定，注意客户端和 APP 同时升级。OTA 现在也不稳定，建议通过线刷刷固件。目测 IRbaby 要再过一两个月才能趋于稳定。现在确实抱歉，给你们带来了不好的体验。

icandura, 2020-08-07:
可是，我固件下载的是最新的版本呢
https://github.com/Caffreyfans/IRbaby-firmware/releases/download/0.4/IRbaby.zip

Caffreyfans, 2020-08-07:
> 可是，我固件下载的是最新的版本呢
> https://github.com/Caffreyfans/IRbaby-firmware/releases/download/0.4/IRbaby.zip

好的，这确实是最新版本的。你反应的问题，这周我会立马更新一次。注意同时升级。

Caffreyfans, 2020-08-10:
现在更新了一下，0.5 版本是最新版，你试试看？

icandura, 2020-08-11:
> 现在更新了一下，0.5 版本是最新版，你试试看？

更新到0.5版本还是不行，设备管理里面是空白的。目前我这边的环境就是固件和软件 0.3 版本是正常使用。
另外问问有没有什么办法完全清除此前记录的Wifi连接信息，
我之前是用一个按键来删除相关的记录信息。
```
//超长按清空Wifi信息并重启
case 3:
	timeClient.end();
	wifiManager.resetSettings();
	delay(500);
	WiFi.disconnect();
	Serial.println("Please Reset Device");
	//ESP.reset();
	break;
```

Caffreyfans, 2020-08-11:
有的，如果你是 nodemcu 的话，D4 引脚接 GND，持续几秒后会清空配置，你会看到蓝灯闪烁一下就行了。

Caffreyfans, 2020-08-12:
最终问题为用户网络问题，请用户在遇到类似情况时应检查自己的网络情况。

closed 2020-08-12 by Caffreyfans