Caffreyfans/IRbaby #25: 在APP中设置引脚时，只能输入数字，无法输入字母，比如D2

gzsjw, 2020-09-07:
是否跟手机型号有关？我的是华为荣耀8X

Caffreyfans, 2020-09-08:
是否跟手机型号有关？
不，是只能输入数字这是 APP 的设定跟手机无关。
字母定义的引脚最终会转换成对应的 gpio 引脚，比如在 nodemcu 上 D5 对应的 gpio 14，那么要用 D5 引脚，就填 14 就行了。
为什么不能用字母定义的引脚？
考虑到固件的通用性，字母对应引脚只针对某些开发板，不同的开发板可能不能，所以这里采用更通用的方式直接填写 gpio 引脚。

gzsjw, 2020-09-08:
多谢了。
发射红外信号时，是不是会有一点光闪一下，我这里没看到有发光，空调也没有响应。
我用的是5mm的红外发射管，还有一种是3mm，不知道跟这个口径有没关系？

gzsjw, 2020-09-09:
请教以下大拿，这个项目是用什么开发环境的，VSCode + PlatformIO IDE 插件吗，能不能调试代码的？

Caffreyfans, 2020-09-09:
> 多谢了。
> 发射红外信号时，是不是会有一点光闪一下，我这里没看到有发光，空调也没有响应。
> 我用的是5mm的红外发射管，还有一种是3mm，不知道跟这个口径有没关系？

没有关系，我用的也是 5mm 二级管，是可以直接驱动的。你用摄像头去拍二级管是能看到闪红光的。

Caffreyfans, 2020-09-09:
> 请教以下大拿，这个项目是用什么开发环境的，VSCode + PlatformIO IDE 插件吗，能不能调试代码的？

可以啊，你可以自己用 pio 编译，很方便的。然后你自己也可以看看串口打印的消息，看看哪里不对。波特率 115200。

gzsjw, 2020-09-09:
把灯都关掉，终于能拍到闪光了；
加了3级管，亮度好像亮一些；
可惜空调还是没反应，估计是码库不支持了；
另外，我观察到mqtt信号，好像是4个一组，第2个是温度，其他是些on\off\auto，大神能否公布下mqtt负载的信息含义？

Caffreyfans, 2020-09-10:
1. 同一空调品牌会有多种编码，如果一种不行再试试另一种，都不行的情况下才能得出该码库无效的结果。
1. MQTT 负载的含义没什么特别，是根据 HomeAssistant [HV/AC MQTT](https://www.home-assistant.io/integrations/climate.mqtt/)插件来定义的。具体含义你可以根据它英文意思推测出来啊，我详细写后面会在 WIkI 上专门吧它列出来。

gzsjw, 2020-09-10:
很高兴你能及时回应，我的空调是海信日立的风管机，有8个编码选择，都试过了没反应；
我再找台美的的试试，是不是格力、美的这些容易接入，日立有点小众，好像连个APP都没有；

Caffreyfans, 2020-09-10:
我不清楚海信日立是两个品牌还是一个品牌，然后你尝试过品牌下所有编码后都没有效果的话就说明码库里确实没有，这没有办法。

closed 2021-06-23 by Caffreyfans