Caffreyfans/IRbaby #46: 录制红外遥控器代码是什么标准的？

yesky84, 2023-09-27:
请问获取到的红外代码是什么标准的？如果要把红外代码添加到esphome来编译需要选择什么标准？是nec,jvc,raw,pronto还是什么？？

Caffreyfans, 2023-10-07:
RAW

yesky84, 2023-10-13:
remote_transmitter:
  pin:
    number: D0
  carrier_duty_percent: 50%
switch:
  - platform: template
    name: LeiNiao Trun on
    turn_on_action:
      - remote_transmitter.transmit_raw:
          code: [4041, -4030, 523, -2030, 519, -2031, 519, -2032, 519, -2031, 520, -1032, 520, -1032, 520, -2031, 520, -1032, 519, -2032, 523, -1029, 519, -2031, 520, -1032, 520, -1032, 520, -1031, 520, -1033, 520, -1032, 519, -2032, 519, -2031, 521, -1032, 519, -2031, 520, -1032, 520, -2031, 521, -1031, 519, -2032, 519, -8370, 4017, -4031, 521, -2029, 524, -2028, 519, -2032, 519, -2032, 520, -1032, 520, -1032, 519, -2032, 519, -1032, 520, -2036, 515, -1032, 520, -2031, 519, -1034, 518, -1032, 520, -1032, 520, -1032, 519, -1033, 520, -2031, 520, -2032, 519, -1031, 520, -2032, 519, -1033, 519, -2031, 520, -1032, 519, -2032, 519, -8344, 4018, -4030, 519, -2033, 518, -2033, 519, -2032, 519, -2032, 518, -1033, 520, -1033, 519, -2031, 520, -1032, 520, -2031,520, -1032, 519, -2032, 520, -1031, 520, -1033, 519, -1032, 520, -1032, 519, -1033, 518, -2032, 520, -2032, 519, -1032, 520, -2032, 518, -1033, 519, -2032, 519, -1033, 519, -2032, 519]
          carrier_frequency: 38kHz

大佬，请看一下。我esphome这样写，为什么有时候可以控制，有时候不行。  观查电视机的指示灯能看到微暗一下，但开不了机。配置差了什么？

Caffreyfans, 2023-11-23:
esphome 我也不熟悉啊